pvlib/pvanalytics #4: Set up continuous integration/delivery

wfvining, 2020-02-20:
- [ ] build with Travis
- [ ] code coverage with coveralls and/or codecov
- [ ] set up lgtm or other linter
- [ ] package and deploy to pypi

wfvining, 2020-02-20:
@wholmgren I was hoping you could give some feedback based on your experience with continuous integration for pvlib-python. Do you have any opinions or advice about the different CI services you have set up, particularly your experience with travis vs. azure pipelines?

wholmgren, 2020-02-20:
I prefer azure for speed and the platforms that it supports. The pvlib python azure config could use some refactoring though - I think @CameronTStark was going to look into that so tagging him here as well.

CameronTStark, 2020-02-20:
Yes, I am in-process for refactoring pvlib python's Azure Pipelines for simplicity. I aim to consolidate our CI as much as possible within Azure Pipelines and forego Travis. We maintain interaction with Travis since it still publishes new versions to PyPI but it would be more ideal to do that with Azure Pipelines. 

I believe the items on your to-do list are capable of being done completely on Azure Pipelines even though you may notice that pvlib python uses the separate apps you listed. I'm unsure what utility might be gained/lost if pvlib python consolidated completely since we're already setup for LGTM/coveralls so doing so is low priority.

Either way I'd be happy to help with an effort to leverage Azure Pipelines here since there are synergies with work on pvlib python.

wfvining, 2020-02-20:
If the goal for pvlib-python is to move as much as possible into Azure Pipelines then I think we should do the same with PVAnalytics for the sake of consistency within the pvlib organization. I'm not wedded to any of the services I listed above, and if many of them can be consolidated into one place that seems like a win to me.

@CameronTStark thanks for offering to help! I will ping you with questions or review requests as I get this spun up.

wholmgren, 2020-02-20:
Most of the services do different things. I was just talking about consolidating the test suite into azure pipelines. 

I’d also check out the new GitHub actions for a new project.

wfvining, 2020-02-20:
I will check out github actions, on first glance it does look fairly promising, and supports the same platforms as Azure Pipelines (linux, macOS, and Windows).

I see your point about different services doing different things, my goal is to have just one service that does lint-build-test-deploy. I have not used LGTM before, but it does look like something worth including in addition to the CI/CD service.

wfvining, 2020-02-25:
I have worked up a GithubAction for linting and testing in #7. 

I think that GithubAcitons is a good way to go since it is in the same place as the code, and (on the surface) seems simpler than Azure Pipelines. I particularly like that adding new workflows does not require any additional setup other than adding a new YAML file.

@wholmgren @CameronTStark If either of you are interested, please take a look and give me some feedback.

wfvining, 2020-03-10:
We now have all these components set up in github actions (and travis).

closed 2020-03-10 by wfvining