pvlib/pvanalytics #11: Publish to pypi when release is tagged

wfvining, 2020-02-27:
Configure github action to build and publish packages on pypi when a release is tagged and all tests have passed.

This can be added as a new job to the "lint and test" workflow using the `needs` key

```yaml
jobs:

  # [...]

  publish: 
    runs-on: ubuntu-latest
    needs: [lint, test]
    if: github.event_name == "release" # ???

    steps:
      # checkout, setup-python, install setuptools and twine, build package, upload package
```

wfvining, 2020-02-27:
Alternatively, we can create a separate workflow that runs only when a release is published

```yaml
on:
  release:
    types: [published]
```

cwhanse, 2020-02-27:
> Alternatively, we can create a separate workflow that runs only when a release is published
> on:
>   release:
>     types: [published]

I don't think we want to publish e.g. pre-release tags to pypi.

wfvining, 2020-02-27:
[Here ](https://help.github.com/en/actions/reference/events-that-trigger-workflows#release-event-release) is the doc for the github `release` trigger.

There is a separate `prereleased` type, so I think the above snippet will only trigger for a non-prerelease, non-draft release. I will do some experimenting elsewhere to be sure though.

wholmgren, 2020-02-27:
FYI we publish pvlib prerelease tags to pypi. It's helpful for debugging and has no cost. https://pypi.org/project/pvlib/#history

prerelease ~ 0.7.0b2
prerelease != untagged

wfvining, 2020-03-03:
Once #12 is closed I will add a workflow to publish releases (including pre-releases). It seems that `pip` (and presumably other tools) will ignore pre-releases when they follow a standard release number convention

https://packaging.python.org/guides/distributing-packages-using-setuptools/#pre-release-versioning

closed 2020-03-05 by wfvining