pvlib/pvanalytics #24: Add shadow detection

wfvining, 2020-03-31:
Should go in the `features` module, I think.

rjstephens, 2020-07-21:
Are you referring to shading of a pyranometer or on the panels?

cwhanse, 2020-07-21:
Both, I think. We want the library to be capable of hosting shadow detection algorithms of various types and uses.

We have an algorithm that can process a time series of irradiance (at ~1min resolution) to label times when the sensor was shaded, which finds shadows that correlate across days (e.g., from nearby poles or wires). It won't reliably identify shadows from soiled sensors. It's a research project to see if, and how, the algorithm can be extended to time-averaged irradiance or power data.

rjstephens, 2020-07-21:
Ok that sounds good. 

Would doing a comparison between the pyranometers at each combination of azimuth and elevation be enough to indicate if a given pyranomter is shaded by a fixed object such as a pole as you mention? I have had success in the past doing this on 15 minute resolution irradiance data.

Do you have a link to any further information on this algorithm and project?

cwhanse, 2020-07-21:
The algorithm I'd like to publish processes one sensor at a time. Having multiple sensors is a different approach. From pvanalytics perspective both techniques are welcome, and we can let users decide which works best for them.

We had a preprint in review that describes the algorithm, then the first author left and I never got it finished. Illustration of the algorithm's output is in this [report](https://www.osti.gov/servlets/purl/1232609), and there is [code](https://github.com/sandialabs/MATLAB_PV_LIB/blob/master/pvl_detect_shadows.m) in PVLib for Matlab.

PR #101 (2021-06-03): Detect shadows from fixed objects in GHI data
## Description

This is a port of the pvl_detect_shadows function from MATLAB_PVLIB.

## Checklist

- [x] Closes #24
- [x] Added new API functions to `docs/api.rst`
- [x] Clearly documented all new API functions with [PEP257](https://www.python.org/dev/peps/pep-0257/) and [numpydoc](https://numpydoc.readthedocs.io/en/latest/format.html) compliant docstrings
- [x] Adds description and name entries in the appropriate "what's new" file 
      in [`docs/whatsnew`](https://github.com/pvlib/pvanalytics/tree/master/docs/whatsnew) 
      for all changes. I [... cut]
closed 2021-06-03 by wfvining