pvlib/pvanalytics #53: Determine azimuth and tilt for fixed PV systems

wfvining, 2020-06-11:
Given a series of power or irradiance measurements determine the azimuth and tilt of the sensor/system.

The PVFleets QA code does this by comparing the solar azimuth at the maximum of parabolas fit to the data for each day and the solar azimuth for the maximum POA irradiance of each day with different tilt and system azimuths. The tilt-azimuth pair that gives the smallest difference is returned as the tilt and azimuth of the system.

wfvining, 2020-06-12:
Here is a slightly more detailed description of the algorithm from the PVFleets QA project.

#### Inputs
* `data` time series of power or irradiance measurements
* `daytime` Boolean time series indicating day (`True`) or night (`False`)
* `sunny` Boolean time series indicating times without clouds (*in the PVFleets implementation this only includes values on days that are sunny for (more or less) the whole day*, it might also make sense to use more fine-grained data - I'm not sure)
* `solar_azimuth` Time series of solar azimuth
* `ghi_clearsky` Time series of clearsky GHI
* `dhi_clearsky` Time series of clearsky DHI
* `dni_clearsky` Time series of clearsky DNI

#### Output
Tuple: `(azimuth, tilt)`

#### Procedure
* get the time for the vertex of a quadratic fit to each day in `data[sunny & daytime]` (an approximation of solar noon)
* get the `solar_azimuth` at these times
* a range of possible system azimuths are determined based on the solar azimuth (at solar noon) during the summer months
* for each possible azimuth and tilt identify the `solar_azimuth` at the maximum clearsky POA irradiance on each day
* find the system azimuth and tilt (values in `[5,10,15,20,25,30]`) where the sum of squared error between the azimuth calculated in the previous step and the azimuth calculated in the second step is minimized

### One questions I have
Can the tilt and azimuth components be separated into different functions, or is it necessary to optimize both variables simultaneously?

cwhanse, 2020-06-15:
Inputs make sense, we can work on the names.

> Can the tilt and azimuth components be separated into different functions, or is it necessary to optimize both variables simultaneously?

There are other methods that try to determine tilt and azimuth separately, for those separate functions make sense. Since this approach relies on the same quadratic fit, I would keep the output as described above.

bt-, 2020-06-16:
@wfvining, would it be possible to setup the API for this function to allow the user to set the resolution of the azimuth and tilt used in the search?

I am reading

>find the system azimuth and tilt (values in [5,10,15,20,25,30])

to mean that it would find the tilt to the nearest 5.  I anticipate this to be a useful function for checking orientation of both sensors and arrays, but I would like to be able to get a tighter estimate if possible.

Are the inputs defined here intended to be user inputs to a function or just general inputs needed for the algorithm?

wfvining, 2020-06-18:
TL;DR: Making the 'resolution' a parameter isn't a problem, but the maximum accuracy of this algorithm is not clear (even with finer resolution). The inputs above are just the general inputs needed for the algorithm, they are probably a subset of the parameters that will be used in the actual implementation.

Making this a parameter wouldn't be a problem; however, I am not sure the algorithm described here will give you extremely high accuracy, even with smaller increments in the candidate tilts and azimuths. @matt14muller, who did the original implementation, has mentioned in passing that there is room to improve the accuracy of this algorithm. It is possible one of the improvements is just finer-grained candidates (as you suggest).

There is one substantial difference between Matt's original algorithm and what I am proposing here that could have bearing on accuracy. In the original algorithm only one year of solar position and clear-sky data are used. Each day in the data is compared to the corresponding day of the year in the clear-sky and solar position. There are some benefits to doing it this way, mainly not having to compute several years of solar position/clear-sky which can be fairly expensive (Matt was running this on several thousand data-sets all of which have more than 2 years of data). I wanted to pass in solar position and clear-sky for the full time series because it matches the rest of the PVAnalytics API and the way things are typically done in pvlib-python. It also provides flexibility in how these parameters are calculated. This change may or may not improve accuracy, but it should not decrease it. The time-resolution of the solar position passed in will have an effect on accuracy, with the best accuracy for one-minute data.

wholmgren, 2020-07-30:
Maybe there is value in implementing whatever PVFleets has done for reference purposes. But what would be really useful is algorithm that uses a minimizer like [`scipy.optimize.minimize`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.minimize.html) and does not require a list of candidate azimuth/tilts or a resolution parameter.

wfvining, 2020-07-31:
I think a benefit of passing a list of candidate azimuths and tilts is that you can very effectively restrict the search space if you already know something about the system. This might save some computation relative to `scipy.optimize.minimize` depending on how many iterations it takes to converge, of course there are some obvious cases where method described here will take many more iterations than an optimizer. 

In general I think you are right, a method that does not require candidate tilts/azimuths would be preferable.

PR #64 (2020-08-24): Infer azimuth and tilt from power or POA irradiance
From scratch re-implementation of the algorithm form PVFleets QA. There are two substantial differences:
* Caller must pass a list of candidate azimuths and tilts. The PVFleets version includes an extra step where the candidate azimuths are chosen based on the data and uses a fixed set of tilts.
* The index of the solar position Series must cover the same time period as the data. In the PVFleets implementation only one year of solar position data is used. This adds some complexity that I think can be avoided by just requiring users pass in so [... cut]
closed 2020-08-24 by wfvining