pvlib/pvanalytics #60: Function to calculate the Variability Index

bt-, 2020-07-07:
I have a function somewhere that calculates the variability index (VI).  I think this package would be a good home for it and I'd be happy to contribute it if there is a place for it.

The features module might be an appropriate place, but a daily float value for VI is returned rather than a boolean.

wfvining, 2020-07-07:
Thanks, @bt-, your contribution would be greatly appreciated! One of our main goals is to provide a home for these kinds of things.

Generally we have been naming things `features.<feature>.<method>` (in this case that would be `features.variability.<bt-s_method>()`) as a way to keep organized and keep space for several different methods to calculate the same feature. The other place that might be good for it is in `metrics`. We don't have anything there yet. What do you think @cwhanse?

cwhanse, 2020-07-07:
In `pvanalytics.metrics`, I think. Thanks @bt-!

PR #106 (2021-05-06): Create function to calculate variability index
## Checklist

- [x] Closes #60
- [x] Added new API functions to `docs/api.rst`
- [x] Clearly documented all new API functions with [PEP257](https://www.python.org/dev/peps/pep-0257/) and [numpydoc](https://numpydoc.readthedocs.io/en/latest/format.html) compliant docstrings
- [x] Adds description and name entries in the appropriate "what's new" file 
      in [`docs/whatsnew`](https://github.com/pvlib/pvanalytics/tree/master/docs/whatsnew) 
      for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` 
      or this Pull Request  [... cut]
closed 2021-05-06 by cwhanse