pvlib/pvanalytics #65: Add theoretical energy production and performance ratio

rjstephens, 2020-07-21:
Is there an intention to add the option for creating a theoretical energy production based on the irradiance and possibly temperature? After that performance ratio could also be calculated.

wfvining, 2020-07-21:
We may want to add performance ratio calculations, but computing theoretical energy production is the domain of [pvlib-python](github.com/pvlib/pvlib-python). `pvlib-python` and `pvanalytics` are split roughly between the modeling PV system performance/produciton and analyzing data from existing PV systems respectively. PVAnalytics leverages the PVLib where needed to support analysis.

I think your suggestion would be in this category: use `pvlib-python` to calculate theoretical energy production and pass the results to a (not yet implemented) PVAnalytics function to compute the performance ration

cwhanse, 2020-07-21:
+1 to adding performance ratio calculation to pvanalytics.

rjstephens, 2020-07-21:
I was referring to the theoretical energy production calculated from just the irradiance and plant capacity (and sometimes temperature) used in the acceptance tests of operational solar farms. The operational yield is then divided by the theoretical yield to get the performance ratio. 

Knowing these values can also help with other aspects of understanding the data quality.

cwhanse, 2020-07-21:
@rjstephens there's a plan to add that "model" to pvlib-python. Any suggestions for a name? The model is

DC power = GHI / 1000 * DC capacity * (1 - temperature coefficient * (T array - T reference)

It's not PVWatts but it's quite similar.

bt-, 2020-07-21:
I've been planning to add performance ratio functions/methods to pvcaptest for quite a while. I have the basic functions written already, but they need to be cleaned up and have proper tests written. I'm open to contributing the basic functions to pvanalytics and then importing them as needed to use in pvcaptest.  Thought I'd suggest that to avoid duplicating our efforts.

I think these would fit in the metrics module?

cwhanse, 2020-07-21:
> I think these would fit in the metrics module?

Yes, they would. +N to avoiding duplication :)

wfvining, 2020-09-04:
#69 added NREL's weather corrected performance ratio. Is that enough to close this out, or should we keep it open to track other performance ratio methods that we might want to add?

rjstephens, 2020-09-12:
Would there be merit in adding a vectorised version of the new function? Much of the work I see being done is in DataFrames and this could offer a significant speed improvement.