pvlib/pvanalytics #70: Enable ReadTheDocs PR builds

kanderso-nrel, 2020-08-26:
pvlib-python and rdtools have configured RTD to automatically build docs as one of the checks on a PR.  I think it's a great feature and use it all the time -- I usually build docs locally for my own PRs, but when reviewing other people's it's very handy to not have to clone the branch and build locally to see how the docs turned out.  Also useful for people who don't necessarily want to learn how to build docs locally.  Any reason not to do the same for pvanalytics?

cwhanse, 2020-08-26:
Would be a great feature to have for pvanalytics. @wfvining something you want to tackle?

wfvining, 2020-08-28:
I can set this up. I originally held off because I wanted to try something with github-actions that would give a bit more feedback on warnings from sphinx ([ammaraskar/sphinx-action](https://github.com/ammaraskar/sphinx-action)).

I don't recall why I was having a hard time setting this up, but I can try again. At the very least I will enable the webhook notification for RTD builds.

wfvining, 2020-08-28:
Since the last time I looked, I think RTD added a "build pull requests" option (before I think you had to opt in to a beta). I have enabled it, so we should start seeing PRs build.

closed 2020-08-28 by wfvining

kanderso-nrel, 2021-12-03:
PR docs builds are happening, but for some reason the status isn't being reported, or at least it isn't in #114.  This page confirms that the build is happening: https://readthedocs.org/projects/pvanalytics/builds/

So it must just be a status reporting issue.  Maybe RTD isn't granted the permissions necessary to post status updates to this repo?  https://docs.readthedocs.io/en/stable/connected-accounts.html#github-permission-troubleshooting

I think I can't access this repo's settings, so someone else will need to check it out if we want to fix this.

kanderso-nrel, 2021-12-03:
Above issue is resolved after hitting the "resync webhook" button on RTD's Admin>Integrations page.  Unclear to me why that would make a difference, but it's working now...