pvlib/pvanalytics #80: `scipy.stats.median_absolute_deviation` is deprecated

wfvining, 2020-10-23:
Getting a warning from the tests for `pvanalytics.quality.outliers`

### From the deprecation warning:

use `median_abs_deviation` instead!
  
  To preserve the existing default behavior, use `scipy.stats.median_abs_deviation(..., scale=1/1.4826)`. The value 1.4826 is not numerically precise for scaling with a normal distribution. For a numerically precise value, use `scipy.stats.median_abs_deviation(..., scale='normal')`.

cwhanse, 2020-10-23:
OK to switch function names from `median_absolute_deviation` to `median_abs_deviation(..., scale='normal')` [here](https://github.com/pvlib/pvanalytics/blob/f6bd37acd1a86c545b2e879b1f5d3d720835c3f3/pvanalytics/quality/outliers.py#L98)

Looks minor to me

wfvining, 2020-10-23:
It should be that simple, but `test_hampel_max_deviation` is failing when I change it.

PR #85 (2020-10-26): Use statsmodels for maximum absolute deviation
Rather than `scipy.stats.maximum_absolute_deviation()` we will use `statsmodels.robust.scale.mad()` to calculate the Maximum Absolute Deviation. This change will let us lower the minimum scipy version to 1.2 `scipy.stats.maximum_absolute_deviation()` was introduces in 1.3 (see #83).

## Checklist

- [x] Closes #80 
- [x] Pull request is nearly complete and ready for detailed review
closed 2020-10-26 by wfvining