pvlib/pvanalytics #96: Handle missing data gracefully in clipping.levels()

wfvining, 2020-11-13:
`features.clipping.levels()` raises some very difficult to decipher value errors form numpy when there are NAs in the data. We should either raise a more meaningful and useful exception or simply deal with NAs by dropping them, applying the clipping filter, reindexing, and filling missing values with `False`.

cwhanse, 2020-11-13:
Can you link an example?

wfvining, 2020-11-13:
Running this:

```python
daytime_power = power.copy()
daytime_power[power < 100] = pd.NA
clip_levels = clipping.levels(daytime_power)
```

Results in `ValueError: autodetected range of [nan, nan] is not finite` from `numpy.histograms`

If you do `clipping.levels(daytime_power.dropna())` it works as expected, but you need to reindex the mask if you want to use it on the original data. My suggestion is to do the `dropna()` and `reindex()` inside `clipping.levels()`.

Alternatively we can add to the documentation that the data must not contain undefined values and raise an exception with a meaningful/useful error message when undefined values are present.

PR #97 (2020-11-17): Ignore missing data in clipping.levels()
## Checklist

- [x] Closes #96 
- [x] Added tests to cover all new or modified code
- [x] Pull request is nearly complete and ready for detailed review
closed 2020-11-17 by wfvining