pvlib/pvanalytics #102: Handling missing data in outliers

camsilva, 2021-03-05:
Hello, first of all very interesting project!

I was trying out the outlier funcions. The `outliers.zscore` function (which depends on `scipy.stats.zscore`) returns a series full of NaNs when at least one NaN is present in data.

Would it be helpful to use the option of ignoring such data? Like `stats.zscore(data, nan_policy='omit')` or dropping NaN values?
Or handling this type of data is not to be contemplated?

cwhanse, 2021-03-05:
> Would it be helpful to use the option of ignoring such data? 

Yes, and thanks for point this out. @wfvining we missed this case in the tests.

wfvining, 2021-03-08:
I think @camsilva's suggestion of exposing the `nan_policy` param makes sense. It does get a little odd though since the returned series is boolean, if you pass `omit` nans will be `False` in the output. The scipy default is `propagate`, but I'm not sure that makes sense here, we probably want `omit` or `raise` as a default.

camsilva, 2021-03-09:
Passing `propagate` is giving me no info on why everything outputs as False. `raise` would need a dropna() before calling but at least is clear on why it behaves like that which is reasonable. `omit` has that odd behavior @wfvining indicates, but maybe nans should not be treated as outliers in the context of these methods (?) and should indeed return `False`.

wfvining, 2021-03-09:
The idea I had for `raise` as the default is that it would alert the user to the fact that they have `nan`s in their data, then they can decide how they want to handle it. In that context we don't need any `dropna()` calls. I would not want to treat nans as outliers, they are easy to find and users may want to handle them differently.

camsilva, 2021-03-09:
Yes, I was mentioning `dropna()` from the user side as a way to overcome the `raise` error. I agree with you!

tgunda, 2021-05-07:
To increase user awareness, we can expose scipys.stats.zscore()’s ‘nan_policy’ parameter as kwargs to the PVAnalytics function. The default ‘raise’ flag will alert users that their data has NaNs so that they can address them as needed, including leveraging scipy. So we will create a PR that exposes the NaNs, takes advantage of scipy API, and ensures returned values are non-NaNs.

PR #108 (2021-05-18): updated outliers.zscore to include nan_policy
## Description

An update to `outliers.zscore` to expose the nan_policy.

## Checklist

The following items must be addressed before the code can be merged. 
Please don't hesitate to ask for help if you are unsure of how to accomplish any of the items. 
*You are free to remove any checklist items that do not apply or add additional items that are 
not on this list*

- [x] Closes #102
- [ ] Added new API functions to `docs/api.rst`
- [ ] Clearly documented all new API functions with [PEP257](https://www.python.org/dev/peps/pep-0257/) and [numpydoc]( [... cut]
closed 2021-05-18 by wfvining