pvlib/pvanalytics #105: Tests for clearsky.reno fail with pvlib 0.8.1

wfvining, 2021-04-06:
PVLib 0.8.1 changed the labeling of clear-sky periods (https://github.com/pvlib/pvlib-python/pull/1074) which is resulting in a test failures in `test_clearsky.py`.

The change in pvlib labels windows on the left by default, while previous behavior used centered labels. We probably want to accept the new default behavior from PVLib (updating the tests to match) and expose the new `align` parameter in `pvanalytics.clearsky.reno`.

cwhanse, 2021-04-06:
> The change in pvlib labels windows on the left by default, while previous behavior used centered labels. 

Other way around; before v0.8.1, pvlib used left labels. pvlib v0.8.1 uses centered labels.

wfvining, 2021-04-06:
> Other way around; before v0.8.1, pvlib used left labels. pvlib v0.8.1 uses centered labels.

Thanks, updated.

wfvining, 2021-04-22:
Actually, the alignment is not exposed in the pvlib API. We should probably just accept the centered labels here, and remove/rewrite tests that relied on left-aligned labels. I'm thinking that my approach to testing this function was flawed to begin with, it might be better to mock the pvlib function and check that the window/scale are computed correctly and passed the pvlib function (otherwise we are spending a bunch of effort testing code that isn't even part of pvanalytics).

cwhanse, 2021-06-04:
Tests for `clearsky.reno` fail because of a bug in `pvlib.clearsky` #https://github.com/pvlib/pvlib-python/issues/1241#

PR #128 (2022-01-27): Get clearsky.reno tests passing again
## Description

The sphinx build in #127 is failing because of #126, and I figured might as well fix #105 while we're at it.  

## Checklist

- [x] Partially addresses #105
- [x] Closes #126
- ~[ ] Added new API functions to `docs/api.rst`~
- ~[ ] Clearly documented all new API functions with [PEP257](https://www.python.org/dev/peps/pep-0257/) and [numpydoc](https://numpydoc.readthedocs.io/en/latest/format.html) compliant docstrings~
- [x] Adds description and name entries in the appropriate "what's new" file 
      in [`docs/whatsnew`](https://github.com/ [... cut]
commit 6fb4c87a (2022-01-27): Get clearsky.reno tests passing again (#128)

* fix #126

* fix #105

* whatsnew
closed 2022-01-27 by kanderso-nrel