pvlib/pvanalytics #107: clearsky algorithm with POA data?

toddkarin, 2021-05-03:
What is the expected behavior of the clear sky algorithm applied to plane-of-array data?

cwhanse, 2021-05-03:
Are you asking about `features.clearsky.reno`?

toddkarin, 2021-05-04:
Yes.

cwhanse, 2021-05-04:
If the clear-sky profile is transposed to the plane-of-array, then I think the algorithm would perform reasonably to label clear-sky periods in plane-of-array data. To be explicit: the plane-of-array data is passed to `ghi`, and the transposed clearsky GHI is passed to `ghi_clearsky`.

The thresholds aren't exposed in this interface.  The thresholds are probably OK for plane-of-array data, ideally at 1 minute. If the irradiance data are at longer intervals, out to 15min, there's evidence that the same thresholds work, but no formal validation.

wholmgren, 2021-05-04:
I did this a few years ago and found the thresholds were difficult to tweak. I had more success converting POA to GHI using `pvlib.irradiance.gti_dirint` and then applying `pvlib.clearsky.detect_clearsky`.