pvlib/pvanalytics #113: Requirements in setup.py are redundant, and partially inconsistent, with requirements.txt

kanderso-nrel, 2021-11-29:
`index.rst` advertises the contents of `requirements.txt` as the package's dependencies:

https://github.com/pvlib/pvanalytics/blob/a3d3860ee848afd5cb7184b36beedcd14dbbdc60/requirements.txt#L1-L6

But the pandas version range specified in `setup.py` is in fact slightly different ([ref](https://github.com/pvlib/pvanalytics/pull/67/files#diff-4d7c51b1efe9043e44439a949dfd92e5827321b34082903477fd04876edb7552R2)):

https://github.com/pvlib/pvanalytics/blob/a3d3860ee848afd5cb7184b36beedcd14dbbdc60/setup.py#L33-L40

Worth considering having a "single source of truth" for the requirement ranges so that there's no way for redundant specs to get out of sync.  I think some packages solve this by including a line in `setup.py` like this: `INSTALL_REQUIRES = open('requirements.txt').read().split('\n')`.  Maybe eliminating `requirements.txt` and keeping version ranges in `setup.py` would make more sense in this case.

Same goes for `DOCS_REQUIRE` and `docs/requirements.txt`.

wfvining, 2021-11-29:
Great suggestion @kanderso-nrel, I've got it on my to-do list to do some general maintenance on dependencies for this project (notably work to support higher versions of pandas) and it would be good to address this as well.

wholmgren, 2021-11-29:
Worth reading this discussion of setup.py vs requirements.txt if you've not already done so: https://caremad.io/posts/2013/07/setup-vs-requirement/ 

It's now 8 years old (a lifetime in python packaging) but so far as I'm aware the advice still applies. TLDR: pvanalytics should not have requirements.txt files

PR #149 (2022-08-15): Delete requirements.txt; related CI & test changes
## Description

There are two main reasons we might want to have a requirements.txt:
- As an environment spec for users.  But pvanalytics is a library, not an application, so we prefer dependency ranges in `setup.py` over dependency versions in a requirements file.
- As an environment spec for testing, if you need specific versions beyond what's specified in setup.py.  We don't (currently) need that, so there's no need for a requirements.txt file either.  The exception is `requirements-min.txt`, which of course is not equivalent to setup.py.

 [... cut]
closed 2022-08-15 by kanderso-nrel