pvlib/pvanalytics #119: Linear Algebra Error thrown in earlier numpy versions when running orientation fixed_nrel() function.

kperrynrel, 2022-01-04:
Following error is thrown when running fixed_nrel() function on a azimuth/tilt estimation pipeline:

raise LinAlgError("SVD did not converge in Linear Least Squares")

LinAlgError: SVD did not converge in Linear Least Squares

This occurred when running numpy=1.19.1. Updated to numpy 1.19.5 and issue was resolved. Need to update minimum numpy installation.

cwhanse, 2022-01-04:
Ah, nvm I see the issue. [This helper function](https://github.com/pvlib/pvanalytics/blob/bb451a030517e0ef180a02d20484261c0897f278/pvanalytics/util/_fit.py#L6) uses `np.polyfit` which is the old numpy API, not yet deprecated AFAIK but less well-mannered than its replaced `np.polynomial`.  We should bring the helper function up to the newer API; transition guide [here](https://numpy.org/doc/stable/reference/routines.polynomials.html).

kperrynrel, 2022-01-04:
I can do that @cwhanse, will cut a PR making an update on that function.

kanderso-nrel, 2022-02-15:
Closed by #120?

kperrynrel, 2022-02-15:
@kanderso-nrel  correct. Closing this--thanks for tracking housekeeping on this!

closed 2022-02-15 by kperrynrel