pvlib/pvanalytics #126: Error in clearsky.reno in newer pandas and numpy versions

kanderso-nrel, 2022-01-26:
I know there are some known pandas issues already, but I think this one is new?  This snippet raises an error for some combinations of pandas and numpy:
```python
from pvanalytics.features.clearsky import reno
import pandas as pd

times = pd.date_range('2019-01-01 12:00', freq='1min', periods=3)
ghi = pd.Series([1,1,1], index=times)
reno(ghi, ghi)  # ValueError: clearsky requires regular time intervals of 15m or less
```

It fails with `numpy==1.21.4` + `pandas==1.3.0`.  It works with `numpy==1.19.5` + `pandas==1.2.5`.

This is the issue:

```python
In [7]: pd.Timedelta("1T") / np.timedelta64(1, "60s")
Out[7]: 60.0  # returns 1.0 on older versions
```

I did not do any digging for changes in either library that may have caused this, or whether the issue can be isolated to a change in just one of pandas or numpy.

cwhanse, 2022-01-26:
This one is new. Probably safer to compute that integer number of minutes with either pandas or numpy but not both. Surprising to see the combination give the wrong answer.

cwhanse, 2022-01-26:
Would that statement work in newer versions as

```
pd.Timedelta("1T") / np.timedelta64(1, "m")
```

Wondering if it's a change in how numpy handles the time unit string.

kanderso-nrel, 2022-01-26:
With the newer versions listed above `pd.Timedelta("1T") / np.timedelta64(1, "m")` gives `1.0`, so perhaps that's a suitable fix.  I tend to agree that sticking with all pandas is likely safer.

PR #128 (2022-01-27): Get clearsky.reno tests passing again
## Description

The sphinx build in #127 is failing because of #126, and I figured might as well fix #105 while we're at it.  

## Checklist

- [x] Partially addresses #105
- [x] Closes #126
- ~[ ] Added new API functions to `docs/api.rst`~
- ~[ ] Clearly documented all new API functions with [PEP257](https://www.python.org/dev/peps/pep-0257/) and [numpydoc](https://numpydoc.readthedocs.io/en/latest/format.html) compliant docstrings~
- [x] Adds description and name entries in the appropriate "what's new" file 
      in [`docs/whatsnew`](https://github.com/ [... cut]
commit 6fb4c87a (2022-01-27): Get clearsky.reno tests passing again (#128)

* fix #126

* fix #105

* whatsnew
closed 2022-01-27 by kanderso-nrel