pvlib/pvanalytics #137: numpy 1.9.X security issue

cwhanse, 2022-02-25:
https://github.com/advisories/GHSA-6p56-wp2h-9hxr

Consider advanced to numpy>=1.21

kanderso-nrel, 2022-03-03:
https://nvd.nist.gov/vuln/detail/CVE-2021-33430 tentatively suggests that this isn't worth worrying about.  It'd be a shame to require such a recent version of numpy (1.21.0 was released Jun 22, 2021) if there's no real need to do so.

kanderso-nrel, 2022-07-06:
I think I'm inclined to do nothing here since the numpy maintainers say this is not a real vulnerability.  See https://github.com/numpy/numpy/issues/18993#issuecomment-1004440986 for discussion among the numpy maintainers regarding this particular security issue as well as others.  I encourage anyone interested to read through that discussion, but my takeaway is that the recent CVEs filed against numpy are spurious and fine for us to ignore.  See also "disputed" here: https://www.cvedetails.com/vulnerability-list/vendor_id-16835/product_id-39445/Numpy-Numpy.html

However, if we do advance to `numpy>=1.21`, we'd need to revise our dependency policy, which currently sticks to NEP-29:

https://github.com/pvlib/pvanalytics/blob/e8ce80154eecabf9088e8c551bd7ebebb79b2e32/docs/index.rst#L75-L82

I don't think it changes our decision here, but maybe of interest regardless, here's a recent unresolved discussion about possibly modifying NEP-29 to accommodate security issues: https://github.com/numpy/numpy/issues/21713.