pvlib/pvanalytics #144: v0.1.2 release planning

kanderso-nrel, 2022-07-05:
It seems prudent to be planning for a release ahead of PVPMC.  The main changes are of course all the gallery examples @kperrynrel has been adding, but I've tagged a few additional things under the [v0.1.2 milestone](https://github.com/pvlib/pvanalytics/milestone/2) I'd like to see get resolved:

- #20
- #112 
- #113

I volunteer to submit PRs for those three issues.  Are there any other things people would like to see in the next release?

cwhanse, 2022-07-05:
Address #137 and #121 perhaps?

kanderso-nrel, 2022-07-06:
Is there a rush for #121?  So long as the older pandas versions aren't causing any problems I don't see a reason to drop them.

cwhanse, 2022-07-06:
No rush and if you aren't concerned about #137, then neither am I.

kperrynrel, 2022-07-06:
TODO on my end: documentation for systems module and orientation module (https://github.com/pvlib/pvanalytics/issues/133)

kanderso-nrel, 2022-08-18:
Looks like #145 is the last thing left on the list before final release cleanup.  It's not critical but might be nice to get in if we're directing PVPMC audience members to our docs.  @cwhanse are you planning to look at that one, or should I go ahead and open a PR to finalize 0.1.2?

cwhanse, 2022-08-18:
Merge #145 when you'd like, I approved.

PR #150 (2022-08-18): v0.1.2 Release

## Checklist

The following items must be addressed before the code can be merged. 
Please don't hesitate to ask for help if you are unsure of how to accomplish any of the items. 
*You are free to remove any checklist items that do not apply or add additional items that are 
not on this list*

- [x] Closes #144
- ~[ ] Added new API functions to `docs/api.rst`~
- ~[ ] Clearly documented all new API functions with [PEP257](https://www.python.org/dev/peps/pep-0257/) and [numpydoc](https://numpydoc.readthedocs.io/en/latest/format.html) compliant docstrings~
- [x] Adds description a [... cut]
closed 2022-08-18 by kanderso-nrel