pvlib/pvanalytics #147: Consider aligning `pvanalytics.system.infer_orientation_fit_pvwatts` with PVWatts better

kanderso-nrel, 2022-07-07:
`infer_orientation_fit_pvwatts` currently uses a basic power model sort of but not entirely resembling PVWatts ([v5](https://pvwatts.nrel.gov/downloads/pvwattsv5.pdf)?).  Some differences include:
- Assuming isotropic diffuse irradiance instead of the Perez transposition model
- Using the Sandia cell temperature model instead of Fuentes
- No IAM model
- Different default temperature coefficient (-0.004 vs -0.0047)
- No "pvwatts losses"

Question to discuss: should these differences be resolved so that it implements the full PVWatts model?  Using `pvlib.modelchain.ModelChain.with_pvwatts` would get it close (there is still a temperature model difference, as of pvlib 0.9.1).  

Either way, as discussed in https://github.com/pvlib/pvanalytics/pull/77#discussion_r511010737, the docstring `Notes` section could use a small revision.

kperrynrel, 2022-12-15:
@cwhanse and @kanderso-nrel I am working this in anticipation of aligning the algorithm better for the benchmarking paper I'm putting together. Here is what I have so far: 

We'd be mucking out pvanalytics.system._power_residuals_from_clearsky() and inserting the new logic. @kanderso-nrel talked about employing ModelChains.with_pvwatts, but he's hesitant to do this as this may significantly slow down the overall function runtime (solar position recalculated at every iteration). Kevin suggested I implement the pipeline piecemeal using the existing PVLib PVWatts functions. To that end we'd be looking at something like the following:

pvlib.irradiance.get_total_irradiance() using the Perez transposition model (model='perez') 
pvlib.iam.physical (IAM model implementation recommended in PVWatts5)
pvlib.temperature.sapm_cell (Sandia cell temperature model) or Fuentes (try both out on validation set and check out results??)
pvlib.pvsystem.pvwatts_dc
pvlib.pvsystem.pvwatts_losses (will need to add this) 
pvlib.inverter.pvwatts 

Before connecting all of these functions together, I wanted to open it up to discussion if there's a better way to do this.

kanderso-nrel, 2022-12-16:
I've done some simple speed comparisons regarding the solar position question and found that ~85-90% of `with_pvwatts().run_model()` runtime is in the solar position and airmass calculations, which for a year of 15-minute data is the difference between completing tens of milliseconds versus hundreds of milliseconds.  The notebook I used is available [here](https://gist.github.com/kanderso-nrel/280c9331d6b469945f68d4f56eb7a399).

In Kirsten's [gallery example](https://pvanalytics.readthedocs.io/en/latest/generated/gallery/system/infer-orientation-fit-pvwatts.html) for this function the optimizer says it evaluated the objective function only 7 times, which is much fewer than I had expected.  I guess we just need to decide whether it's worth accepting a runtime of a few seconds in exchange for simpler implementation.  I suspect the answer is probably "yes".

cwhanse, 2022-12-16:
@kanderso-nrel I think a 10x speed-up is of significant value here, at the cost of a more complex implementation. If I follow your thinking, the complexity would be 'behind the curtain' for users (calculate solar position once and preserve it in the model chain). I think one likely use of this function is to check orientation for each of a large number of systems.

cwhanse, 2022-12-16:
@kperrynrel thumbs down for me on using the Fuentes temperature model. I know that's what's in PVWattsv5 so if we are hewing to v5 over other concerns, I'll yield. But PVWatts v5 is the only use of Fuentes I know of, and I suspect it's quite a bit slower than any of the pvlib alternatives.

kanderso-nrel, 2022-12-16:
> the complexity would be 'behind the curtain' for users

Yes, the complexity would be in code development and maintenance on our side: either manually chaining together the pvwatts model functions as @kperrynrel described, or employing a kludgy workaround in ModelChain as I did in that notebook.  My initial reaction to that `HackedLocation` class was that it was unsuitable to use in real code, but maybe that's unnecessarily puritan.

> Fuentes... quite a bit slower than any of the pvlib alternatives.

+1, and if the recommended guidance is to restrict to clear-sky conditions then the main advantage of Fuentes (thermal transience) is largely irrelevant anyway.

PR #180 (2023-01-20): Updated to PVWatts5 methodology in the pvwatts orientation estimation function system.infer_orientation_fit_pvwatts()
Update the PVWatts calculations to better align with PVWatts 5 methodology. 

- [x] Closes #147 
- [x] Added tests to cover all new or modified code.
- [x] Clearly documented all new API functions with [PEP257](https://www.python.org/dev/peps/pep-0257/) and [numpydoc](https://numpydoc.readthedocs.io/en/latest/format.html) compliant docstrings.
- [x] Added new API functions to `docs/api.rst`.
- [x] Non-API functions clearly documented with docstrings or comments as necessary.
-  [... cut]
closed 2023-01-20 by kanderso-nrel