pvlib/pvanalytics #156: Stale data function check runs slow

kperrynrel, 2022-09-09:
@kanderso-nrel and @cwhanse I'm working on integrating some of the pvanalytics functions into the Fleets QA pipeline, and I noticed that the PVanalytics implementation for stale detection is incredibly slow compared to the original pandas version that I put together: 

`import pvanalytics
import time
import pandas as pd
import matplotlib.pyplot as plt

def stuck_data(time_series):
    '''
    When 4 or more consecutive values in the series are identical, when rounded
    to the third decimal place, these consecutive values are flagged as 
    stuck data
    
    Parameters
    ----------
    time_series: a series of power or irradiance data but can be applied
    to other series that are consecutive data
    
    Returns
    ----------    
    stuck_data_mask :
        Boolean time series mask identifying values as stuck, False.
    '''
    time_series = time_series.round(decimals=3)
    #Count consecutive values in the sequence using Pandas functionality
    consecutive_val_counts = time_series.groupby((time_series != time_series.shift()).cumsum()).transform('size') 
    #If 4 or more consecutive values, flag it 
    stuck_data_mask = ~(consecutive_val_counts < 4).astype(bool)
    return stuck_data_mask  

# Time how long it takes to run each equation
# Read in a time series
file = "C:/Users/kperry/Documents/source/repos/pvanalytics/pvanalytics/data/ac_power_inv_2173_stale_data.csv"
data = pd.read_csv(file, index_col=0, parse_dates=True)
data = data.asfreq("15T")
data['value_normalized'].plot()

start = time.time()
stuck_mask = pvanalytics.quality.gaps.stale_values_round(data['value_normalized'],
                                                          window=4)
end = time.time()
print("PVAnalytics time execution")
print(end - start)

data['value_normalized'].plot()
data.loc[stuck_mask, "value_normalized"].plot(ls='', marker='.')
plt.legend(labels=["AC Power", "Detected Stale Data"])
plt.xlabel("Date")
plt.ylabel("Normalized AC Power")
plt.title("PVAnalytics Stuck Data mask")
plt.tight_layout()
plt.show()

start = time.time()
stuck_mask_local = stuck_data(data['value_normalized'])
end = time.time()
print("Local time execution")
print(end - start)

data['value_normalized'].plot()
data.loc[stuck_mask_local, "value_normalized"].plot(ls='', marker='.')
plt.legend(labels=["AC Power", "Detected Stale Data"])
plt.xlabel("Date")
plt.ylabel("Normalized AC Power")
plt.title("Local Stuck Data mask")
plt.tight_layout()
plt.show()`

Running the above outputs the following:

[image]

With very similar results (my function just counts all stale values in a block as stale, unlike PVAnalytics, which keeps the first stale value):

[image]

[image] 

There is obviously a pretty serious change in the runtime for essentially the same function. This difference became really pronounced when I was running hundreds of time series in sequence. I'd recommend we speed this up using some kind of pandas functionality, as it's vectorized.

cwhanse, 2022-09-09:
I'm all for speeding that up; it's probably the use of `.apply(<custom function>)` which is just a loop behind the curtain.

I think we could discuss if the first value (of a sequence of approximately equal values) should be considered stale, or not. In my view, it's not stale - it is different from the preceding value. Maybe we can have an option to label or not label the first value. I'm guessing that labeling all values, then un-labeling the first in each window, would be almost as fast as your code.

AdamRJensen, 2022-09-09:
Hmm I seem to be getting the opposite results (@kperrynrel perhaps you can see if there is something obvious wrong in my code)

[image]

kperrynrel, 2022-09-09:
@AdamRJensen I think it's the units on your snapshot--the local stuck_data() function is returning in microseconds, whereas the pvanalytics.quality.gaps.stale_values_round() is in milliseconds, so the local stuck_data() is over 300x faster, if I got the units right here

kperrynrel, 2022-09-09:
@cwhanse I think we can fully replicate what we have in terms of results from the PVAnalytics function to this local function (i.e. keep first value and don't count it as stale), and get that same level of speedup. Let me see what I can do in terms of retooling.

PR #158 (2022-11-04): Stale data speedup
## Description

*Thank you for your contribution! Please provide a brief description of the problem and the proposed solution or new feature (if not already fully described in a linked issue)*

This PR speeds up the stale_data_round() function, by replacing the rolling + apply routine with a vectorized pandas routine.

## Checklist

The following items must be addressed before the code can be merged. 
Please don't hesitate to ask for help if you are unsure of how to accomplish any of the items. 
*You are free to remove any checklist items that do not apply or add additional  [... cut]
closed 2022-11-04 by kperrynrel