pvlib/pvanalytics #169: Configure the linter to only check lines of code that change for new PR's

kperrynrel, 2022-11-22:
Right now I have a PR that is failing from a previously pushed PR pep8 error (see irradiance module in https://github.com/pvlib/pvanalytics/pull/162). This PEP8 issue is deliberate as it errors on URL text in the docstring that we want to keep. @kanderso-nrel mentioned pvlib-python overrides previous PR issues in main (that are deliberate) when making new PR's by configuring the linter to only check lines of code that changed.

wholmgren, 2022-11-23:
[flake8 deprecated the diff option](https://flake8.pycqa.org/en/latest/release-notes/5.0.0.html#deprecations). For new/young projects the recommendation is to configure the linter to fail on all errors. It's much more robust that checking the diff.

kanderso-nrel, 2022-11-28:
I think adding a few `noqa` comments is fine for now.  If someday it does make sense to switch to only linting the diff (which I agree is complicated), the suggestions here are a place to start: https://github.com/PyCQA/flake8/issues/1760#issue-1463492509

closed 2022-11-28 by kanderso-nrel