pvlib/pvanalytics #175: v0.1.3 planning

kanderso-nrel, 2022-12-09:
Shall we aim for a release next week?  It's been ~3 months since 0.1.2 and we've got some nice things waiting to go out ([whatsnew](https://github.com/pvlib/pvanalytics/blob/main/docs/whatsnew/0.1.3.rst)).

@kperrynrel what are your thoughts on #146?

@abhisheksparikh I think we'd be happy to have #167 if it's ready in time, but also no big deal to push it to the next release if you don't have bandwidth available right now. 

If anyone has additional items they'd like to get into 0.1.3 (thermopile offsets?), feel free to add them to the [milestone](https://github.com/pvlib/pvanalytics/milestone/3) and solicit reviews.

abhisheksparikh, 2022-12-09:
Yes, I want to complete the GHI and POA quality checks in #167. I will get the GHI check in by 12/11.

kperrynrel, 2022-12-09:
@kanderso-nrel I just went over #146  again and saw it's only doing an example for the pvwatts orientation method, which I'm currently more comfortable with than the infer_orientation_daily_peak() method (Matt's original implementation is yielding different results from the pvanalytics implementation, which is less accurate). I'm okay with picking this back up here today and getting it squared away for the next version release. I'll need to eventually retool infer_orientation_daily_peak() to get it more accurate/faster.

PR #177 (2022-12-16): v0.1.3 release
- [x] Closes #175
- ~[ ] Added tests to cover all new or modified code.~
- ~[ ] Clearly documented all new API functions with [PEP257](https://www.python.org/dev/peps/pep-0257/) and [numpydoc](https://numpydoc.readthedocs.io/en/latest/format.html) compliant docstrings.~
- ~[ ] Added new API functions to `docs/api.rst`.~
- ~[ ] Non-API functions clearly documented with docstrings or comments as necessary.~
- [x] Adds description and name entries in the appropriate "what's new" file 
      in [`docs/whatsnew`](https://github.com/pvlib/pvanalytics/tree/master/docs/whatsnew) 
       [... cut]
closed 2022-12-16 by kanderso-nrel