pvlib/pvanalytics #176: Consider switching sphinx themes

kanderso-nrel, 2022-12-14:
Our docs currently use the `alabaster` theme, resulting in a website which I (very subjectively) find a little difficult to read and navigate.  Compare with docs built with other themes:

- pvanalytics (alabaster): https://pvanalytics.readthedocs.io/en/stable/whatsnew/0.1.2.html
- pvlib (pydata-sphinx-theme): https://pvlib-python.readthedocs.io/en/latest/whatsnew.html
- rdtools (rtd-sphinx-theme): https://rdtools.readthedocs.io/en/stable/changelog.html
- twoaxistracking (sphinx_book_theme): https://twoaxistracking.readthedocs.io/en/stable/

I'd like to switch to another theme, but I don't have a strong opinion on what new theme to choose.  Some considerations:

Themes affect not just font and styling but also navigation.  For example, the pydata-sphinx-theme separates out top-level navigation to a top navbar instead of putting the entire website tree in the left sidebar.  See examples of many themes here: https://sphinx-themes.org/#themes

The three `pvlib/*` projects with sphinx docs currently use three different themes.  Using the same theme across multiple projects might be helpful in that experience using a theme in one project would transfer to another.  It may also help with "branding" the pvlib family, if that's something we are interested in.  One downside of using the same theme for all projects is that it might be harder to know which project you're looking at if all of our docs look the same.

cwhanse, 2022-12-14:
I somewhat prefer the new pvlib-python theme (pydata) over the current pvanalytics theme (alabaster). I don't like the rtd-sphinx theme; perhaps because that's what pvlib-python just left.

I note that both numpy and scipy use the pydata theme and their home documentation pages look very similar. In fact, both suggest that we're putting too much content on the home documentation page.

bt-, 2022-12-14:
I’ve been finding the latest pvlib docs very pleasant to navigate and I
think it would be helpful to have consistent themes. Not sure if anyone
references it other than me, but I’m open to changing pvcaptest to match.

PR #178 (2023-01-20): Switch docs to pydata-sphinx-theme
- [x] Closes #176
- ~[ ] Added tests to cover all new or modified code.~
- ~[ ] Clearly documented all new API functions with [PEP257](https://www.python.org/dev/peps/pep-0257/) and [numpydoc](https://numpydoc.readthedocs.io/en/latest/format.html) compliant docstrings.~
- ~[ ] Added new API functions to `docs/api.rst`.~
- ~[ ] Non-API functions clearly documented with docstrings or comments as necessary.~
- [x] Adds description and name entries in the appropriate "what's new" file 
      in [`docs/whatsnew`](https://github.com/pvlib/pvanalytics/tree/master/do [... cut]
closed 2023-01-20 by kanderso-nrel