pvlib/pvanalytics #184: Soiling detection methods

cwhanse, 2023-02-01:
Here's an interesting approach: https://doi.org/10.1016/j.isci.2021.102165

cwhanse, 2023-04-24:
And another http://dx.doi.org/10.1109/JPHOTOV.2023.3243719

abhisheksparikh, 2023-05-29:
> Here's an interesting approach: https://doi.org/10.1016/j.isci.2021.102165

Since we have the soiling module in pvlib, should this be in pvlib-python? Also, the SRR method's code has a github repository - I wonder if Michael would be interested in getting it in pvanalytics (or pvlib).

kandersolar, 2023-05-30:
> Since we have the soiling module in pvlib, should this be in pvlib-python?

IMHO: methods of predicting soiling ratio for performance modeling purposes belong in pvlib-python and methods of detecting and/or extracting soiling from measured performance belong elsewhere (perhaps here in pvanalytics).

> Also, the SRR method's code has a github repository - I wonder if Michael would be interested in getting it in pvanalytics (or pvlib).

Maybe you are talking about https://github.com/NREL/pv_soiling?  That repository is not actively developed, but SRR is also implemented in RdTools: [RTD](https://rdtools.readthedocs.io/en/stable/generated/rdtools.soiling.soiling_srr.html) and [GH](https://github.com/nrel/rdtools/blob/master/rdtools/soiling.py).  Since there's already a maintained (and better) version elsewhere, I don't see much value in having a duplicate implementation of it here.  And I very much doubt Mike et al. would be interested in transferring the "official" version from RdTools to pvanalytics.

abhisheksparikh, 2023-05-30:
> > Since we have the soiling module in pvlib, should this be in pvlib-python?
> 
> IMHO: methods of predicting soiling ratio for performance modeling purposes belong in pvlib-python and methods of detecting and/or extracting soiling from measured performance belong elsewhere (perhaps here in pvanalytics).

Yes, I agree - but then, what method should we use/explore to detect soiling from the performance data? I am referring to Table 6. of [https://doi.org/10.1016/j.isci.2021.102165](https://doi.org/10.1016/j.isci.2021.102165). 

> Maybe you are talking about https://github.com/NREL/pv_soiling? That repository is not actively developed, but SRR is also implemented in RdTools: [...]

Okay, that makes sense.