pvlib/pvanalytics #188: Account for lower GHI limit in check_irradiance_consistency_qcrad

AdamRJensen, 2023-03-10:
I'm surprised that the [check_irradiance_consistency_qcrad](https://pvanalytics.readthedocs.io/en/stable/generated/pvanalytics.quality.irradiance.check_irradiance_consistency_qcrad.html) function does not take into account the 50 W/m^2 lower limit as specified in the reference. For GHI values less than the 50 W/m^2 limit the test cannot be applied (see reference below).

At the very least I suggest that a comment on this is added to the note as otherwise we cannot really claim to use the criteria in the reference.

**Reference:**
[image]

AdamRJensen, 2023-03-15:
> I think that flagging low irradiance at nighttime is OK in the context of pvanalytics. The package provides other functions which label day/night periods, and each quality function returns a time series of Boolean for the critieria it checks. The temptation is to regard 'True' here as equivalent to a valid measurement - and that is not always the case, as the discussion brings out.
_Originally posted by @cwhanse in https://github.com/pvlib/pvanalytics/issues/167#issuecomment-1375753120_

I think this is a very valid point, thus, perhaps we should go beyond labeling flags as binary. In particular, it could be considered to use nans for data points that are outside the test criteria. For example, typically values that are less than 50 W/m^2 can't be tested, and labeling them 0 (test failed) or 1 (test passed) is not correct. In such cases, setting the flag as nan would be a reasonable option IMHO. In fact, this was the method followed in the this [study](https://www.researchgate.net/publication/355309934_Expert_quality_control_of_solar_radiation_ground_data_sets).

adriesse, 2023-03-16:
> perhaps we should go beyond labeling flags as binary.

I fully agree.

cwhanse, 2023-03-16:
I would agree if, in the implementation, nan doesn't introduce spurious True values when AND-ing series of flags.

```
a = np.array([np.nan, True, False])
b = np.array([np.nan, True, True])
np.logical_and(a, b)

array([ True,  True, False])
```

adriesse, 2023-03-16:
I suggest all integer values.  No nans.  Negative values could be algorithm exception codes and non-negative values diagnostic output, for example.

cwhanse, 2023-03-16:
I'm a hard no on returning a Series of integer codes in place of a Series of booleans, since these filtering functions are meant to define filters for data. We could return a second Series of codes indicating state, in addition to returning the Series of boolean, and let the user decide which states to filter on.

cwhanse, 2023-03-16:
@AdamRJensen I think that the lower bound on GHI (50 W/m2) is applied in the consistency check in this [line](https://github.com/pvlib/pvanalytics/blob/369dedaadc9a694ab9967a4312f230b562770d57/pvanalytics/quality/irradiance.py#L278) and is sourced from this [dict](https://github.com/pvlib/pvanalytics/blob/369dedaadc9a694ab9967a4312f230b562770d57/pvanalytics/quality/irradiance.py#L19).

AdamRJensen, 2023-03-23:
> @AdamRJensen I think that the lower bound on GHI (50 W/m2) is applied in the consistency check in this [...]

Huh! Well I guess I'm just in disagreement with how the check is implemented then :) The way I interpret the 50 W/m^2 limit is that the checks aren't reliable in assessing if the measurement is incorrect in that region, hence the values below 50 W/m^2 are considered passing (and should be set to True and not False).

closed 2023-05-30 by AdamRJensen