pvlib/pvanalytics #201: Power, irradiance, and temperature examples from PV Fleets pipeline?

kperrynrel, 2024-01-09:
Hey @kandersolar and @cwhanse since we have the logic for time shifts now merged into main, I wanted to ask if we'd be interested in adding some example notebooks illustrating the QA process that we've been using in Fleets (for temperature, irradiance, and power/energy streams):

https://github.com/kperrynrel/fleets_qa_examples/tree/main 

I know we want to keep this repo pipeline-agnostic, so these would just act as a couple of examples which illustrate how our group is using PVAnalytics functions in our work. I still need to integrate the new time shift logic into these examples (it's literally plugging in the new functions), but these could be just jupyter notebook examples, or I can integrate them as examples into Sphinx. No worries if we're not interested, but as I'm pointing multiple people towards this example repo, it could be valuable to include. Let me know your thoughts!

cwhanse, 2024-01-09:
@kperrynrel those would be welcomed. The base layer of functions is agnostic to workflow but we envisioned adding a layer above with various pipelines, similar to the ModelChain concept in pvlib.

kperrynrel, 2024-01-09:
Awesome! Would we want to mimic Pvlib formatting, and add these as jupyter notebooks under a /tutorials/ folder, as so? 
[image]
I can start working this PR here this week once we make a decision on where/how we'd want to include these.

cwhanse, 2024-01-09:
I think we'd rather have examples as .py files as in the [pvlib example](https://github.com/pvlib/pvlib-python/tree/main/docs/examples) gallery, but I'll defer to @kandersolar here.

kandersolar, 2024-01-10:
Yes, .py files are much preferred over .ipynb.  A new folder in the [examples](https://github.com/pvlib/pvanalytics/tree/main/docs/examples) to create a new section in the [gallery](https://pvanalytics.readthedocs.io/en/latest/generated/gallery/index.html) sounds good to me.

kperrynrel, 2024-01-11:
Sounds good @kandersolar and @cwhanse, I will start working up a PR for this now. Thanks!

PR #202 (2024-02-12): Added PV Fleets QA pipeline examples

- [x] Closes #201 
- [x] Non-API functions clearly documented with docstrings or comments as necessary.
- [x] Adds description and name entries in the appropriate "what's new" file 
      in [`docs/whatsnew`](https://github.com/pvlib/pvanalytics/tree/main/docs/whatsnew) 
      for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` 
      or this Pull Request with `` :pull:`num` ``. Includes contributor name 
      and/or GitHub username (link with `` :ghuser:`user` ``).
- [x] Pull request is nearly complete and ready for detailed review [... cut]
closed 2024-02-12 by kandersolar